chengxuxin/extreme-parkour #1: about reindex() and go1 support

rendashuai17, 2023-09-28:
Thanks for your contribution! Really appreciate you open source this work. I notice you reindex the dof orders when getting the observation, which has not been used in the original ETH code, what`s the reason of this?  I directly change the URDF to new_go1, and tested with the weights you provided, but the robot performs like the dof is in a wrong order, I wander is this related to the reindex operation?

chengxuxin, 2023-09-28:
Thanks for your issue. The reindex function is mainly used to match the dof oder on hardware since unitree uses slightly different ordering. I have checked to use new_go1.urdf with 051-40 and there is no issue.
 

https://github.com/chengxuxin/extreme-parkour/assets/37524252/645c5d75-32ba-4469-a212-d2d0838af00b

Are you running with 051-41 or 051-42? Policies with depth images won't work out of the box because if you don't change the camera position, the camera on go1 will be inside the robot so the depth is always black. You can try to move the camera to a proper location. see [here](https://github.com/chengxuxin/extreme-parkour/blob/671a8a0f033eb6f43a7f54c4efcdda368797fd06/legged_gym/legged_gym/envs/base/legged_robot_config.py#L95).

https://github.com/chengxuxin/extreme-parkour/assets/37524252/517ce8b9-0da5-47ed-a97d-8c4fe4f013d7

rendashuai17, 2023-09-28:
Thanks for your reply! The problem is just as you say. I change the camera position to [0.3,0,0.03] and the view is no longer black, now I want to train a depth distillation policy with this new camera position on go1, (I use 051-40 as the base policy ) however, the depth loss is not declining, is there any problem?
 the loss is as follows:
[image]
my training command is this: 
[image]

chengxuxin, 2023-09-29:
Make sure the camera position is appropriate so the robot is able to see the obstacles in front. You can also try without direction distillation, which is easier to train.

closed 2023-10-28 by chengxuxin

hakieh, 2024-01-27:
hi, I am also interesting in training the policy with go1, so have you successfully train the policy with go1 ? Can you share the appropriate camera coordinates? Thanks a lot!

zrhrongrong666, 2024-05-28:
Hello, have you deployed successfully? If a RealsenseD435 camera is used and the depth coordinate system of the camera is centered on the left camera, is the installation training position of the camera the left center position relative to the robot's center of mass position?