chengxuxin/extreme-parkour #2: go1 learns 3 legged policy

drewjenks01, 2023-09-29:
Hello, when I try training the base policy for go2 ```python train.py --exptid 054-01 --task go1``` it learns to only use the front 2 legs and the back left leg. I tried training the a1 policy to see if it happens but that turned out fine and then I tried training go1 again and it learned 3 legs again. Is this something with the go1 config?

chengxuxin, 2023-10-02:
Go1 support is not added yet, so training with `--task go1` will have unexpected behaviors especially the terrains contain extreme cases since the max torque for go1 is 23.7Nm vs A1~33 Nm. You can try to change the urdf [here](https://github.com/chengxuxin/extreme-parkour/blob/4ea4d0f88b43dcdc6926254906826592345fc39f/legged_gym/legged_gym/envs/a1/a1_parkour_config.py#L64) instead of using `--task go1` and see what happens.

dstx123, 2023-10-31:
I am also interested in training the go1 policy. May I ask that have you successfully deployed this algorithm after modifying the urdf? By the way, the go1 urdf file in Isaac Sim seems to be the old version.

RudeGoose, 2023-12-12:
some problem with you. When I try to train the base policy for other quals, it always learns to use the front leg and the back left leg. I tried to lower the difficulty of the terrain and reduce the rew_feet_edge, but they don't work. May I ask you that have you successfully deploy the alogorithm or fix the problem?

ruanruan0313, 2024-08-01:
> some problem with you. When I try to train the base policy for other quals, it always learns to use the front leg and the back left leg. I tried to lower the difficulty of the terrain and reduce the [...]
I also have this problem，May I ask you have solved this problem？

chrisy9, 2026-11-16:
Useful practices of mine (on a proprietary quadruped, not certain to work universally)
- adjust terrain geometry hyperparameters in accordance with the robot
- implement a per-dof pose reward (see mjlab)
- do not introduce action delay too early

Good luck

dms-rd, 2026-11-16:
I have deployed policy on Go1. I have trained policy with modification in reward values and action delay. This works best for me.