chengxuxin/extreme-parkour #7: guide on running on unitree a1

nhooram, 2023-10-11:
Great work. I've been dreaming about such a control algorithm being open sourced for running on the A1 since I got one a few years ago. It would be really cool if there was a guide on how to get this to run on the actual A1 robot.

tenhearts, 2023-10-11:
Thanks for liking our project. We will add hardware guide and code soon.

1242713693, 2024-02-29:
Hello, have you migrated your code to A1? Hope to get your reply

traxu001, 2025-03-28:
> Hello, have you migrated your code to A1? Hope to get your reply

hi ！have you deploy your code on A1 now？

1242713693, 2025-03-28:
没有

cc
***@***.***

        

         原始邮件

发件时间：2025年3月28日 21:22
收件人：chengxuxin/extreme-parkour ***@***.***&gt;
抄送：CC ***@***.***&gt;, Comment ***@***.***&gt;