chengxuxin/extreme-parkour #22: Collection Time of per iteration

ChengEeee, 2024-01-09:
[image]
Why does the  project's Collection Time of per iteration take several times more time than the LeggedGym's.
Hope your reply！Thank you very much

ChengEeee, 2024-01-09:
LeggedGym's Collection Time of per iteration less than 1s

1242713693, 2024-02-28:
Hello, is your display graphical interface in VScode? My computer doesn't have a GPU, can I run it directly on the CPU? If you run directly on the CPU, can you visualize, like in your video! thank you.

chengxuxin, 2024-02-28:
> ![image](https://private-user-images.githubusercontent.com/94172651/295166283-ce59beea-245f-4589-9675-bf98d6927329.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Please provide more info on what commands and gpu you are using.

chengxuxin, 2024-02-28:
> Hello, is your display graphical interface in VScode? My computer doesn't have a GPU, can I run it directly on the CPU? If you run directly on the CPU, can you visualize, like in your video! thank you.

Yes you can train teacher and student without gui. To visualize on a headless machine please add `--web` arg to visualize in vscode.

1242713693, 2024-02-28:
> Yes you can train teacher and student without gui. To visualize on a headless machine please add `--web` arg to visualize in vscode.

I have installed Live Preview in VScode according to the prompt, and the code can run normally, but there is no interface. I do not know whether it is my wrong operation or other problems. I sincerely hope to get your help, thank you
[image]

chengxuxin, 2024-02-28:
The `train.py` won't give you rendering. You have to run `python play.py --exptid the_model_you_trained --web` to get a render of your policy. Please see the readme carefully.

1242713693, 2024-02-28:
> The `train.py` won't give you rendering. You have to run `python play.py --exptid the_model_you_trained --web` to get a render of your policy. Please see the readme carefully.

Thank you very much for your help. Do you mean that I can only watch in vscode through the command --web after the training is completed? If the model is being trained, can it also be viewed via --web?
There is another thing that needs to trouble you, if I configure the GPU for the computer, basically, which model of the ordinary Nvidia graphics card can make the code run smoothly? thank you.

chengxuxin, 2024-02-28:
you can run `play` while you are running `train`, no problem. I recommend at least 16 gb gpu memory, best would be 3090 4090.