chengxuxin/extreme-parkour #24: KeyError: 'depth_actor_state_dict'

XanderWangXY, 2024-01-19:
When I use command "python save_jit.py --exptid xxx-xx" I meet this error.

Traceback (most recent call last):
  File "/home/adc/pycharm-community-2023.3.2/plugins/python-ce/helpers/pydev/pydevd.py", line 1534, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/adc/pycharm-community-2023.3.2/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/adc/extreme-parkour/legged_gym/legged_gym/scripts/save_jit.py", line 123, in <module>
    play(args)
  File "/home/adc/extreme-parkour/legged_gym/legged_gym/scripts/save_jit.py", line 91, in play
    policy.actor.load_state_dict(ac_state_dict['depth_actor_state_dict'], strict=True)
KeyError: 'depth_actor_state_dict'

How can I solve it?

XanderWangXY, 2024-01-19:
I have solved it..

XanderWangXY, 2024-01-24:
> I am getting the same error. Can you share how you solved it?

change "args.exptid" to the same file name as the trained model file

Indiasuperman, 2024-07-01:
Have you deploy the model into real robot?