chengxuxin/extreme-parkour #26: The Loss_depth/depth_encoder is 0 when i train distillation policy

LargeNumberZZ, 2024-01-28:
[image]
Hi! Thanks for you great job! And I want to know that is it normal for this loss to be 0 all the time？

BaiHiAN, 2025-06-15:
> [image] Hi! Thanks for you great job! And I want to know that is it normal for this loss to be 0 all the time？

Hello, may I ask approximately how much the depth_actor_loss you trained will converge to in the end?

sky-98k, 2026-02-09:
I also noticed this problem. In the file on_policy_runner.py, the author sets depth_encoder_loss = 0. I don't understand why, but obviously, this is not correct.