chengxuxin/extreme-parkour #27: How to deploy on a real robot

hakieh, 2024-01-29:
Hello, I am very interested in your work. I would like to know how this code can be deployed to a real robot, could you introduce how to deploy it?

1242713693, 2024-02-29:
Hello, have you migrated your code to A1? Hope to get your reply

xyyl, 2024-03-12:
Hi @chengxuxin ,
I have read your paper about this repo, important achievements have been made. I am very interest in your work. I am faced with difficulties when trying to deploy the repo in a Unitree A1, I have figured out all components of obs_buf(in a real robot) which corresponds to legged_robot.py/obs_buf(in simulation) but these items:{(self.env_class != 17).float()[:, None], (self.env_class == 17).float()[:, None],(self.contact_filt.float()-0.5)}. could you help me? Thank you!

KurokoWD, 2024-06-03:
@xyyl Have you ever solved this problem?I also want to know how to get {(self.env_class != 17).float()[:, None], (self.env_class == 17).float()[:, None],(self.contact_filt.float()-0.5),self.delta_next_yaw} in real robot

hanlinniu, 2024-08-12:
Hi guys, have you installed issacgym successfully on a jetson nx board on A1 robot?