chengxuxin/extreme-parkour #29: After running, you can't see the simulation interface

1242713693, 2024-02-26:
[image]
How to solve this problem?

chengxuxin, 2024-02-28:
Please clarify the issue to be better assisted.

hdShang, 2024-02-28:
> Please clarify the issue to be better assisted.

[image]
Can not see this

hdShang, 2024-02-28:
> Please clarify the issue to be better assisted.

[image]
Can not see this interface.

hdShang, 2024-02-29:
> ![image](https://private-user-images.githubusercontent.com/133180305/307710198-6f640716-fb78-4d82-a9c9-6c3adf43c4a2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Comment out line 43 in train.py