chengxuxin/extreme-parkour #31: Questions about step5 in README.md file

XanderWangXY, 2024-02-29:
In README.md file: 
5. Save models for deployment:
```bash
python save_jit.py --exptid xxx-xx
```
After I finish this step, how can I deploy the model to a real robot?

chengxuxin, 2024-02-29:
Unfortunately, we do not open source our hardware code as of now. But it is very straight forward to implement based on other opens source implementations, such as 
https://github.com/yxyang/cajun
https://github.com/yxyang/fast_and_efficient
https://github.com/Improbable-AI/walk-these-ways

XanderWangXY, 2024-03-01:
> Unfortunately, we do not open source our hardware code as of now. But it is very straight forward to implement based on other opens source implementations, such as https://github.com/yxyang/cajun https://github.com/yxyang/fast_and_efficient https://github.com/Improbable-AI/walk-these-ways

Thank you very much

1242713693, 2024-03-01:
> 不幸的是，我们目前还没有开源我们的硬件代码。但基于其他开源实现来实现是非常直接的，比如 [https://github.com/yxyang/cajun ](https://github.com/yxyang/cajun) [https://github.com/yxyang/fast_and_efficient ](https://github.com/yxyang/fast_and_efficient) [https://github.com/Improbable-AI /走这些路](https://github.com/Improbable-AI/walk-these-ways)

Looking forward to seeing you open source your hardware code on github soon, I think it will be cool

1242713693, 2024-03-01:
> > 不幸的是，我们目前还没有我们开源的硬件代码。但基于其他开源实现来实现是非常直接的，比如[https://github.com/yxyang/cajun ](https://github.com/yxyang/cajun) [https://github.com/yxyang/fast_and_efficient ](https://github.com/yxyang/fast_and_efficient) [https ://github.com/Improbable-AI/走这些路](https://github.com/Improbable-AI/walk-these-ways)
> 
> 非常感谢
Look forward to deploying this open source code to A1