chengxuxin/extreme-parkour #34: A1 部署问题

Noietch, 2024-03-18:
大家看到这个issue的可以一起讨论一下A1的部署问题吗？

Noietch, 2024-03-18:
深度相机的输入尺寸是多大，训练的时候original = (106, 60)然后resized = (87, 58)，D435的输入远大于这个尺寸这个是怎么取

Noietch, 2024-03-19:
关于部署上的问题：
参考walk these ways的部署方案，需要给obs_buf，但是本仓库里的get_obs()这个函数和walk these ways里的有很大的不同，如何将两者进行对应是这个问题，之前的issue也提到了在compute_observations这个函数中
```python
obs_buf = torch.cat((#skill_vector, 
                            self.base_ang_vel  * self.obs_scales.ang_vel,   #[1,3]
                            imu_obs,    #[1,2]
                            0*self.delta_yaw[:, None], 
                            self.delta_yaw[:, None],
                            self.delta_next_yaw[:, None],
                            0*self.commands[:, 0:2], 
                            self.commands[:, 0:1],  #[1,1]
                            (self.env_class != 17).float()[:, None], 
                            (self.env_class == 17).float()[:, None],
                            self.reindex((self.dof_pos - self.default_dof_pos_all) * self.obs_scales.dof_pos),
                            self.reindex(self.dof_vel * self.obs_scales.dof_vel),
                            self.reindex(self.action_history_buf[:, -1]),
                            self.reindex_feet(self.contact_filt.float()-0.5),
                            ),dim=-1)
        priv_explicit = torch.cat((self.base_lin_vel * self.obs_scales.lin_vel,
                                   0 * self.base_lin_vel,
                                   0 * self.base_lin_vel), dim=-1)
        priv_latent = torch.cat((
            self.mass_params_tensor,
            self.friction_coeffs_tensor,
            self.motor_strength[0] - 1, 
            self.motor_strength[1] - 1
        ), dim=-1)
        if self.cfg.terrain.measure_heights:
            heights = torch.clip(self.root_states[:, 2].unsqueeze(1) - 0.3 - self.measured_heights, -1, 1.)
            self.obs_buf = torch.cat([obs_buf, heights, priv_explicit, priv_latent, self.obs_history_buf.view(self.num_envs, -1)], dim=-1)
        else:
            self.obs_buf = torch.cat([obs_buf, priv_explicit, priv_latent, self.obs_history_buf.view(self.num_envs, -1)], dim=-1)
        obs_buf[:, 6:8] = 0  # mask yaw in proprioceptive history
```
这个为什么要reindex不是很理解

XanderWangXY, 2024-04-29:
Hi! I'm also trying to deploy it on my robot. how is your work going now? Do you know what these parameters mean?

[image]
They relate to the input dimension of the neural network. I find them in "save_jit.py" and "legged_robot_config.py"

yhykid, 2024-05-21:
请问您的实物代码是参考哪个仓库的呢？？

XanderWangXY, 2024-05-21:
> 请问您的实物代码是参考哪个仓库的呢？？

我目前还没有在机器人上部署过强化学习项目，请问你有没有哪些成功部署的实物代码项目呢，可以交流一下吗

zrhrongrong666, 2024-05-28:
你好，请问部署成功了吗？如果使用RealsenseD435摄像机，其位置的安装如何安装？代码中angle[-5,5]代表什么?单位是什么呢？
可不可以交流一下，十分感谢！

Noietch, 2024-05-29:
> 你好，请问部署成功了吗？如果使用RealsenseD435摄像机，其位置的安装如何安装？代码中angle[-5,5]代表什么?单位是什么呢？ 可不可以交流一下，十分感谢！

A1 应该是自带相机的，angle[-5,5]单位看起来应该是角度，他这个工作部署上面非常困难，issue基本上都是关于部署的，但是作者可能focus其他工作了，所以我没有继续follow

yhykid, 2024-06-03:
在ICRA会场问了可心，实际部署是track方向

zrhrongrong666, 2024-06-04:
> 请问obs_buf中的delta_yaw是如何计算的，在实际部署中如何生成cur_goals?

delta_yaw 应该是表示当前朝向与目标朝向之间的角度差,实际部署中应该会结合相机，还没有将这个部署过程完成，很抱歉无法详细回答。

KurokoWD, 2024-06-04:
@yhykid 你好，请问能详细解释下track方向是什么意思吗，是指部署时通过joystick给出参考方向吗？

yhykid, 2024-06-04:
应该是这个意思的 @KurokoWD

1242713693, 2024-07-16:
有大佬部署成功了吗？

tenhearts, 2024-07-31:
> 关于深度信息的问题： （1）部署时深度相机的输入尺寸是多大，训练的时候original = (106, 60)然后resized = (87, 58)，D435的输入远大于这个尺寸这个是怎么取，在youtube视频里目测的fov应该不是很大，所以可能是进行了crop操作，再进行resize，而不是直接resize （2）在保存jit的时候为什么深度信息那个网络是单独保存的而不是一起保存，且使用save_jit.py保存出来的网络无法在play.py里面使用，这个play.py里的似乎是调试代码

1) Deploy的时候确实是直接resize到(87, 58)的
2) 因为vision部分的频率是10hz，action部分的频率是50hz的，真实部署的时候也是单独两个进程运行两个网络，之间进行交流的，所以两个网络需要单独保存

tenhearts, 2024-07-31:
> 你好，请问部署成功了吗？如果使用RealsenseD435摄像机，其位置的安装如何安装？代码中angle[-5,5]代表什么?单位是什么呢？ 可不可以交流一下，十分感谢！

A1上自带realsense d435相机，angle[-5, 5]指pitch angle

1242713693, 2024-08-01:
有大佬知道这个问题如何解决吗？
[image]

hanlinniu, 2024-08-12:
你们有在A1的Jetson NX 板子上成功安装IssacGym吗？

2641405707, 2024-11-24:
> 大家看到这个issue的可以一起讨论一下A1的部署问题吗？几乎所有的issue都指向部署

请问你成功部署在A1上了吗，可以交流一下吗？

heartInsert, 2025-01-13:
已经2025年了，有人成功了？兄弟们别🦌了

traxu001, 2025-03-28:
> 已经2025年了，有人成功了？兄弟们别🦌了

哥们，部署成功了吗？

heartInsert, 2025-03-28:
@traxu001 换go2了，用walk-these-ways

traxu001, 2025-03-28:
> [@traxu001](https://github.com/traxu001) 换go2了，用walk-these-ways

校友哎，可以和您加个联系方式，请教一下吗，谢谢老哥

heartInsert, 2025-03-28:
@traxu001 什么鬼，校友，你哪个学校的

traxu001, 2025-03-28:
> [@traxu001](https://github.com/traxu001) 什么鬼，校友，你哪个学校的

SZU呀，机电的，如果可以，这是我的wechat：Hanxuy1336，谢谢老哥

heartInsert, 2025-03-28:
@traxu001   QQ方便吗，微信用得少  ，  QQ： 361846570

nhooram, 2025-04-15:
@heartInsert  @tenhearts @hanlinniu @Noietch @traxu001  Dear Chinese brothers, has anyone been successful? Please share your ways with the rest of the world 🤜🤛

change-every, 2025-04-23:
I have deployed this repo on the Unitree Go2, which can climb some obstacles. However, the performance does not fully match the result in the paper. I believe further communication could be beneficial. @heartInsert @nhooram @traxu001

nhooram, 2025-04-23:
> I have deployed this repo on the Unitree Go2, which can climb some obstacles. However, the performance does not fully match the result in the paper. I believe further communication could be beneficial. @heartInsert @nhooram @traxu001 

Would you be so kind to write up a short document about how to deploy the trained weights to the real hardware?

change-every, 2025-04-23:
Sure, I’ll open-source my deployment code later as a reference. Hopefully, with everyone’s help, we can get the performance up to the level shown in the paper.

hanlinniu, 2025-04-24:
> Sure, I’ll open-source my deployment code later as a reference. Hopefully, with everyone’s help, we can get the performance up to the level shown in the paper.

Hi, I am also deploying this on my Go2 robot, but found the robot didn't perform very well, if you could open source your code, please let me know, and hope we can improve it together.

change-every, 2025-04-28:
Hi, guys. I have open-sourced my deployment code. Check it out and feel free to contact me with any questions or suggestions. I believe we can improve it together.

BaiHiAN, 2025-06-11:
> Hi, guys. I have open-sourced my deployment code. Check it out and feel free to contact me with any questions or suggestions. I believe we can improve it together.

还有就是，你当时复现的时候action_loss和yaw_loss大概是多少呢？

BaiHiAN, 2026-03-06:
> > 2. ion部分的频率是10hz，action部分的频率是50hz的，真实部署的时候也是单独两个进程运行两个网络，之间进行交流的，所以两个网络需要单独保存
> 
> hello，请问你部署之后，能跳跃gap吗？我的只能跳高就是step，不能跳远，好像识别不出来gap。。。
我在mujoco中试过，也跳跃不了gap；还有就是想请教一下，为什么在isaacgym中normalize之前会有depth_image = depth_image * -1呢？是因为相机z轴默认是反方向？