chengxuxin/extreme-parkour #36: Run error

1242713693, 2024-04-08:
According to the operation steps, the system reported an error, I have tried many ways, can not solve, hope you can get help, thank you very much
[image]

asibatian, 2024-06-25:
Here are the methods that might help:
- set `headless=True`
- reduce `num_envs`
- reduce num of terrains

This seems a problem that related with GPU memory. You can check the GPU memory status using `nvidia-smi` in another terminal.

song-ruiyang, 2024-11-28:
I have also encountered this problem and tried mang ways but failed to solve it.May I ask if you have solved it now?