chengxuxin/extreme-parkour #38: The system reported an error after 7K training

1242713693, 2024-04-13:
The system reported an error after 7K training
[图片]
Hope to get your reply, thank you

1242713693, 2024-04-13:
There is an error in distillation, hope to get your help, thank you
[image]

tenhearts, 2024-07-31:
It is cuda out of memory error. You should try device with large memory space or decrease the num of environments.

1242713693, 2024-08-11:
> It is cuda out of memory error. You should try device with large memory space or decrease the num of environments.

Hello, I would like to ask, have you successfully deployed this code on A1? thank you

tenhearts, 2024-08-12:
> > It is cuda out of memory error. You should try device with large memory space or decrease the num of environments.
> 
> Hello, I would like to ask, have you successfully deployed this code on A1? thank you

I am the first co-author...

1242713693, 2024-08-13:
> > > 这是cuda内存不足错误。您应该尝试使用具有大内存空间的设备或减少环境的数量。
> > 
> > 
> > 您好，我想问一下，您是否已在 A1 上成功部署了此代码？谢谢
> 
> 我是第一合著者...

Have you succeeded in physically porting it?